Woolfrey/software_robot_library #14: Consistency of Link Indexing

Woolfrey, 2022-01-16:
At the moment link indexing doesn't quite match up when computing the inverse dynamics:

```
com = this->fkChain[i]*this->link[i+1].get_com();

I = this->fkChain[i].rotation()
   *this->link[i+1].get_inertia()
   *this->fkChain[i].rotation().transpose();

m = this->link[i+1].get_mass();
```
Is there are way we can change this so it's more logically consistent / less confusing?

Maybe we can have some sort of `baseLink` object which is separate from `fkChain`.

ssutjipto, 2022-01-17:
It is possible to decrease the `std::vector<Link> link` object by having a `baseLink` variable in the class so that `link` and `fkChain` line up. 

The alternative is that to include the `baseTf` as `fkChain[0]`, and doing this should increase the size of `fkChain` to align with `link`. I think I prefer this since keeping the baseLink parameters in `link` means that we don't have to handle the case of a separate variable (with duplicate code/additional if-statements). It also makes it easier to loop through if it was ever necessary. It could also possibly clean up the `update_forward_kinematics()` function. 

What do you think?

Woolfrey, 2022-01-19:
The second idea sounds good. Busy at work right now so let me think about it more clearly when I get time!

Woolfrey, 2022-01-26:
I was thinking we could split the `Link` class in to a `RigidBody` class and a `Joint` class. We can treat the base of the robot as a `RigidBody` and then have the `SerialLink` class connect all the `RigidBody` objects via the `Joint` objects.

This would also make the code more flexible going forward for mobile manipulation, multi-limb robots, etc.

I originally followed Peter Corke's structure with `Link` encompassing everything, but it's another of his paradigms that I want to move away from.

What do you think?

closed 2022-02-01 by Woolfrey