Woolfrey/software_robot_library #18: Forward kinematics chain incomplete

ssutjipto, 2022-02-03:
Currently the final entry of `fkChain` represents the pose of last joint ("right_J6") w.r.t. the `base` link (assuming there is no tool attached to the robot end-effector). 

If the `endpointTF` is set as the identity, the Jacobian returned from calling `SerialLink::get_jacobian()` will use the pose of the last joint ("right_j6") rather than the pose of the "right_hand"  which represents the "stump" of the robot. 
If the `endpointTF` is a valid 4x4 homogenous transform then this endpoint will be _added_ to "right_j6" instead of "right_hand", which would alter the subsequent kinematic and dynamic parameters calculated. 

I'm not sure how you'd like to structure the addition of the "right_hand" so that the `fkChain` can store poses from the whole chain. It seems that it could be added either as an additional link or stored as another `RigidBody` similar to the `base` object in SerialLink. And depending on what you think is the best way to structure it we'll most likely need to update this function `SerialLink::update_forward_kinematics()`. 

Also, in the URDF the value of "right_hand" represents both a link and a joint. It's labelled as a "fixed" joint and not a "revolute" joint, so once there's a parser it can be possible to distinguish between the rigid bodies and varying types of joints. Maybe this might influence your decision?

After we sort this out and update the `SerialLink::get_jacobian()` function so that it uses the `i` which represents the endpoint in `fkChain[i]`, then #2 should be completed.

Woolfrey, 2022-02-03:
Hi @ssutjipto, this is what I was referring to in our meeting the other day. Let me explain with a picture:

[image]

The first pose in fkChain uses the pose of the first link then the joint transform:
```
this->fkChain[0] = this->base.get_pose()*this->joint[0].get_pose(this->q[0]);
```
Then every transform up the the last _joint_ uses only the joint poses:
```
for(int i = 1; i < this->n; i++)
{
	this->fkChain[i] = this->fkChain[i-1]*this->joint[i].get_pose(this->q[i]);
}
```
To get to the physical end-effector itself, I was thinking to use the pose of another `RigidBody` object:
```
this->fkChain[this->n] = this->fkChain[this->n-1]*this->link[this->n-1].get_pose()*this->endpointTF;
```
So in effect, the kinematic chain is book-ended by two `RigidBody` objects with a bunch of `Joint` objects connecting them in between. So when solving the dynamics, it should be easy to line up `fkChain[i]` (an `Isometry3f` object) with `link[i]` (a `RigidBody` obect).

The `Eigen::Isometry3f endpointTF` object is only intended as an additional offset made by the user, like the frame of a mounted camera, a tool attached to the end, etc. It would also make it easy to alter the endpoint online if you're picking up and moving objects around. In most cases I can imagine it is just left as the identity matrix.

Assuming we follow the proposed format, then the decision is how to assign the final link transform, either as a `RigidBody` object or a "fixed" `Joint` object. Is it easy to combine the link and fixed joint of the  _right_hand_ in the urdf to a `RigidBody`? Or should we extend the `Joint` array to have a fixed joint ourselves?

Woolfrey, 2022-02-03:
Also, when you create an Issue, you can add a _Project_ so it will appear as a ticket in the projects list.

I've added the relevant labels too.

ssutjipto, 2022-02-07:
Thanks for adding it the relevant labels and to the project. 

After some discussion, the issue was resolved by adding the transform between the final joint to the endpoint of the robot (stump) (i.e. "link 3 TF" if referring to the image posted above by @Woolfrey) to the `RigidBody` object `link[this->n-1]` where `n` is the number of joints the robot possesses. 

See the following: b18128f1a27e1fedd5c9263a6754ff3811019f15

closed 2022-02-07 by ssutjipto