Woolfrey/software_robot_library #53: Combine Kinematics & Dynamics Calculations

Woolfrey, 2022-04-15:
At the moment we have two separate functions for computing the forward kinematics chain, and another for the inverse dynamics.

I think we can put both of these together in to a single function, maybe inside `set_joint_state()`. This way we don't need to run through 2 separate loops to compute all the necessary kinematic and dynamic properties. It will also make it easier to analyse and debug code since it's all located in the one place.

We could probably put an option to turn the dynamics on and off. This way we can save on computation time if dynamic control is not needed.

Woolfrey, 2022-04-20:
Hey @ssutjipto, I put `update_forward_kinematics()` and `update_inverse_dynamics()` inside of `set_joint_state()`.

A quick visual inspection of the output when running `./robot_library_test` looks correct. Can you confirm?

If everything is OK you can delete `update_forward_kinematics()` and `update_inverse_dynamics()`.

ssutjipto, 2022-08-28:
Everything looks good here! I've checked the results against the output from KDL.
55166cfeb50e42af29e735ee950bad521e44df95

closed 2022-08-28 by ssutjipto