Woolfrey/software_robot_library #57: Can't compile config branch

Woolfrey, 2022-07-14:
Hey @ssutjipto , here's what happens when I try and build:
```
woolfrey@woolfrey-SB2:~/workspace/RobotLibrary/Test$ cmake ../
-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error: File /home/woolfrey/workspace/RobotLibrary/cmake/RobotLibraryConfig.cmake.in does not exist.
CMake Error at /usr/share/cmake-3.16/Modules/CMakePackageConfigHelpers.cmake:330 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:47 (configure_package_config_file)

-- The C compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring incomplete, errors occurred!
See also "/home/woolfrey/workspace/RobotLibrary/Test/CMakeFiles/CMakeOutput.log".
woolfrey@woolfrey-SB2:~/workspace/RobotLibrary/Test$ 
```
so I've got no idea ¯\_(ツ)_/¯

ssutjipto, 2022-07-15:
Sorry, I missed a file when I committed earlier. I've just pushed the change so it should work now! Let me know if you encounter any other issues. 
fdcde299d89debc44cac67a9f831ce629209b8b6

Woolfrey, 2022-07-16:
I've got it to compile, and when I build a ros package with `catkin_make` it works. But I can't seem to declare a class in a cpp file.

In my CMakeFile for the ros package:
```
add_executable(polynomial src/polynomial.cpp)
add_dependencies(polynomial ${trajectory_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
target_link_libraries(polynomial ${catkin_LIBRARIES} RobotLibrary::Trajectory)
```
In the cpp file:
```
#include <iostream>
#include <Polynomial.h>

int main(int argc, char *argv[])
{
	
	Polynomial trajectory;
	
	std::cout << "This is a test." << std::endl;
	
	return 0;                                                                                  // No problems with main
}
```
It appears to be throwing a linking error?
```
/usr/bin/ld: CMakeFiles/polynomial.dir/src/polynomial.cpp.o: in function `main':
polynomial.cpp:(.text+0x2b): undefined reference to `Polynomial::Polynomial()'
```
But the namespace set for RobotLibrary in the CMakeFile is:
```
  NAMESPACE
    RobotLibrary::
```
And when I try ```RobotLibrary::Polynomial``` I get:
```
/home/woolfrey/catkin_ws/src/trajectory/src/polynomial.cpp:13:2: error: ‘RobotLibrary’ has not been declared
   13 |  RobotLibrary::Polynomial trajectory;
      |  ^~~~~~~~~~~~
```

Woolfrey, 2022-07-16:
@ssutjipto

Woolfrey, 2022-07-17:
@ssutjipto Okay, so, I can compile a ROS package with the SerialLink class, but **not** the Polynomial class?!

I looked over the CMakeLists in RobotLibrary but can't see any difference between Model and Trajectory.

Package is here: https://github.com/Woolfrey/trajectory/tree/devel

ssutjipto, 2022-07-21:
@Woolfrey There's no matching constructor in the Polynomial class to be able to call `Polynomial trajectory` in polynomial.cpp

If you want to be able to create the object like that then in Polynomial.h, I can change ln 18 from this `Polynomial();` to `Polynomial() { }`. But this depends on how you want it to be used. 

If this solves your problem you can close the issue!

Woolfrey, 2022-07-22:
OMG I can't believe it was something so simple...!

Anyway, it compiled. I'll merge the branches and we can continue.

closed 2022-07-22 by Woolfrey