Woolfrey/software_robot_library #91: Merge `template` branch in to `master`

Woolfrey, 2023-08-24:
I've completed the fundamental forward kinematics and inverse dynamics for branching structures.

I would like to merge it in to the `master` branch, delete the `kinematic_tree` and `template` branches, and continue working from `devel` as standard.

@ssutjipto, when you have time can you approve that the calculations / code is correct?

ssutjipto, 2024-04-01:
I noticed that there were some errors with the dynamics calcluations. 

The numbers were way off e.g. the torques due to gravity when `q = VectorXf::Zero(numJoints)` were:
- `[0         0   9.34867    -13.7102    26.0236    -0.707591   16.3492     -0.517681]` 
- `[0         0   -39.1958   -6.19528   -11.6481     2.13474     -0.702464  -0.0273044]` (correct)

I narrowed this down to the link's global centre of value mass being incorrect. I realised that the value wasn't being updated in the `Link::update_state` function.

I modified the function to update the global centre of mass there.  See commit 8131efb.

After changing this, the dynamics calculations align with the old values that were calculated. So, I'm happy for this to be merged. 

@Woolfrey I'm not sure if this is the modification you would have made so if you want to change it differently feel free to, but If you're happy with the change then please merge! :smile:

Woolfrey, 2024-04-02:
Hey @ssutjipto, thanks!

You should have posted results earlier and I could have solved it sooner!

We should make a check list / flow chart for assessing the inverse dynamics in future. Maybe even an automated script that pinpoints the problem would be useful.

Woolfrey, 2024-04-03:
Merged to `master` branch (renamed from `main` for consistency).

Deleted old branches:
- `devel`
- `kinematic_tree`
- `template`

Created new branch `devel_control` for working on control classes.

Going forward it might be a good idea to create specific branches for working on specific modules (e.g. `devel_control`, `devel_model` etc.)

closed 2024-04-03 by Woolfrey

ssutjipto, 2024-04-04:
Thanks for merging! 

I should have, but I didn't get a chance to really look at the problem till that day!! 

I've always wanted to add tests to this so that it wasn't manual but I haven't had the time yet :cry: