Woolfrey/software_robot_library #103: FIx indenting of comments

Woolfrey, 2024-04-08:
The indenting for comments is out of alignment after code is pushed and pulled again. We need to go through and rectify the spacing in all the files.

:bangbang: @ssutjipto I have recently set `gedit` to insert 5 spaces instead of an indent when using the `tab` button, so you should configure your IDE to do the same.

FYI my current style is:
```
function()
{
     5 space indenting for new scope         // Comments beginning at 100th column on right hand side (yes I know this is not exactly 100 spaces across, it just doesn't fit neatly in the tiny github window)
}
```
Here's the list:
- [ ] Math
     - [ ] /include/Math.h
- [ ] Model
     - [ ] /include/Joint.h
     - [ ] /include/KinematicTree.h
     - [ ] /include/Link.h
     - [ ] /include/Pose.h
     - [ ] /include/RigidBody.h
     - [ ] /test/urdf_test.cpp
- [ ] Trajectory
     - [ ] /include/CartesianTrajectory.h
     - [ ] /include/MultiTrapezoid.h
     - [ ] /include/Polynomial.h
     - [ ] /include/Spline.h
     - [ ] /include/TrajectoryBase.h
     - [ ] /include/TrapezoidalVelocity.h
     - [ ] /include/Waypoints/h
     - [ ] /test/multi_trapezoid_test.cpp
     - [ ] /test/polynomial_test.cpp
     - [ ] /test/spline_test.cpp
     - [ ] /test/trapezoidal_test.cpp

Woolfrey, 2024-04-17:
@ssutjipto, I fixed the indenting on my end and pushed code. It looks correct to me when viewing in github. Can you confirm and close this issue?

ssutjipto, 2024-05-18:
@Woolfrey 
On my side it still seems to be a mix of both tabs and spaces still. In the image (showing KinematicTree.h), the line character represents a tab and the dots are spaces. This is through all the files. 

[image]

Are you sure about 5 spaces? I think 2 or 4 is more common. Let me know and I can help make the changes.

Woolfrey, 2024-05-19:
I had to use Find & Replace for all tabs and substitute in 5 spaces, so I think it should be OK now. I'll push the code to the `control` branch which I've been working on.

ssutjipto, 2024-05-19:
@Woolfrey 
just letting you know the screenshot is not from the control branch, it's from devel since it's one of the branches where I could find your commits for fixing up the indenting.

Since this repository is public, I feel that it's better to pick an indentation equal to a more common spacing amount that people use. Otherwise it might be annoying for people to work with the repository.
Google C++ Style guide says 2, but people argue that it can be difficult to see the indents. So maybe, 4 is better. But I can't seem to find anyone using 5, this is really a rare. The next number up that people use from 4 is 8, but some people find 8 excessive.

ssutjipto, 2024-06-02:
Here is a short list of example repositories within the robotics space that follow 2 or 4 spacing for indentation. 

2 spaces:
- pcl (Point Cloud Library)
- moveit 
- Google projects generally (https://google.github.io/styleguide/)
- ROS2
- eigen
- octomap
- ceres-solver

4 spaces:
- opencv 
- realsense-ros 
- orocos_kdl
- ORB_SLAM2
- AirSim (Microsoft simulator for drones, cars, more)

Also, style is more than just the tab width so some of these repositories have their own style guide and they sometimes have a clang-format file that the IDE uses to adhere to the desired formatting. 

I really can't find any repository (not limited to the robotics repositories) that uses an odd number for indentation.

I realise that you've changed most of these files to 5 spaces from tabs/4 spaces. But I can change this once we've settled on a number to use.

Woolfrey, 2024-06-02:
The `tab` I used was whatever the default was in `gedit`. It may have been 5 spaces, I don't remember. Either way, I think using 4 spaces is nicer than using 2 spaces:
```
class Haiku
{
    public:
     
        Haiku() {}
     
        static std::string read_line(const unsigned int &lineNumber)
        {
            switch(lineNumber)
            {
                case 1: return "Worker bees can leave.\n";
                case 2: return "Even drones can fly away.\n";
                case 3: return "The Queen is their slave.\n";
                    
                default: throw std::invalid_argument("[FLAGRANT ERROR] Expected a number between 1 and 3, "
                                                     "but you input " + std::to_string(lineNumber) + ".");
            }
        }
          
        static std::string read()
        {
            std::string haiku;                                                                      // Value to be returned
               
            for(int lineNumber = 1; lineNumber < 4; lineNumber++)
            {
                haiku += read_line(lineNumber);
            }
               
            return haiku;
        }   
};
```

2 spaces:
```
class Haiku
{
  public:
     
    Haiku() {}
     
    static std::string read_line(const unsigned int &lineNumber)
    {
      switch(lineNumber)
      {
        case 1: return "Worker bees can leave.\n";
        case 2: return "Even drones can fly away.\n";
        case 3: return "The Queen is their slave.\n";
                    
        default: throw std::invalid_argument("[FLAGRANT ERROR] Expected a number between 1 and 3, "
                                             "but you input " + std::to_string(lineNumber) + ".");
      }
    }
          
    static std::string read()
    {
      std::string haiku;                                                                            // Value to be returned
               
      for(int lineNumber = 1; lineNumber < 4; lineNumber++)
      {
        haiku += read_line(lineNumber);
      }
               
      return haiku;
    }   
};
```

I find 2 spaces very difficult to discern different lines of code when I'm scanning for something particular.

Woolfrey, 2025-05-06:
This has been completed (I think?).

Either way, the tab width for me is set to 4 spaces in my gedit:

[image]

closed 2025-05-06 by Woolfrey