Woolfrey/software_robot_library #109: List requirements

Woolfrey, 2024-04-19:
Someone I was collaborating with was unable to compile the library on Ubuntu 20.04.

We need to check the compatibility and list it in the `README.md` file for future users.

Woolfrey, 2024-07-08:
@ssutjipto you mentioned this in the last meeting?

anubhav-dogra, 2024-07-12:
Hi @Woolfrey 
This is my terminal output after the command `sudo make install`. I think the issue is related to the version of eigen library. 
```
Scanning dependencies of target Trajectory
[  7%] Building CXX object Trajectory/CMakeFiles/Trajectory.dir/test/polynomial_test.cpp.o
In file included from /noetic/software_robot_library/Trajectory/include/Polynomial.h:12,
                 from /noetic/software_robot_library/Trajectory/test/polynomial_test.cpp:10:
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:20:9: error: ‘Vector’ in namespace ‘Eigen’ does not name a template type; did you mean ‘VectorXd’?
   20 |  Eigen::Vector<DataType,Eigen::Dynamic> position;
      |         ^~~~~~
      |         VectorXd
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:21:9: error: ‘Vector’ in namespace ‘Eigen’ does not name a template type; did you mean ‘VectorXd’?
   21 |  Eigen::Vector<DataType,Eigen::Dynamic> velocity;
      |         ^~~~~~
      |         VectorXd
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:22:9: error: ‘Vector’ in namespace ‘Eigen’ does not name a template type; did you mean ‘VectorXd’?
   22 |  Eigen::Vector<DataType,Eigen::Dynamic> acceleration;
      |         ^~~~~~
      |         VectorXd
In file included from /noetic/software_robot_library/Trajectory/include/Polynomial.h:12,
                 from /noetic/software_robot_library/Trajectory/test/polynomial_test.cpp:10:
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:40:31: error: ‘Vector’ in namespace ‘Eigen’ does not name a template type; did you mean ‘VectorXd’?
   40 |   TrajectoryBase(const Eigen::Vector<DataType,Eigen::Dynamic> &startPoint,
      |                               ^~~~~~
      |                               VectorXd
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:40:37: error: expected ‘,’ or ‘...’ before ‘<’ token
   40 |   TrajectoryBase(const Eigen::Vector<DataType,Eigen::Dynamic> &startPoint,
      |                                     ^
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:51:10: error: ‘Vector’ in namespace ‘Eigen’ does not name a template type; did you mean ‘VectorXd’?
   51 |   Eigen::Vector<DataType,Eigen::Dynamic> query_position(const DataType &time)
      |          ^~~~~~
      |          VectorXd
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:80:10: error: ‘Vector’ in namespace ‘Eigen’ does not name a template type; did you mean ‘VectorXd’?
   80 |   Eigen::Vector<DataType,Eigen::Dynamic> _startPoint;
      |          ^~~~~~
      |          VectorXd
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:82:10: error: ‘Vector’ in namespace ‘Eigen’ does not name a template type; did you mean ‘VectorXd’?
   82 |   Eigen::Vector<DataType,Eigen::Dynamic> _endPoint;
      |          ^~~~~~
      |          VectorXd
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:90:55: error: ‘Vector’ in namespace ‘Eigen’ does not name a template type; did you mean ‘VectorXd’?
   90 | TrajectoryBase<DataType>::TrajectoryBase(const Eigen::Vector<DataType,Eigen::Dynamic> &startPoint,
      |                                                       ^~~~~~
      |                                                       VectorXd
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:90:61: error: expected ‘,’ or ‘...’ before ‘<’ token
   90 | TrajectoryBase<DataType>::TrajectoryBase(const Eigen::Vector<DataType,Eigen::Dynamic> &startPoint,
      |                                                             ^
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h: In constructor ‘TrajectoryBase<DataType>::TrajectoryBase(int)’:
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:96:42: error: class ‘TrajectoryBase<DataType>’ does not have any field named ‘_startPoint’
   96 |                                          _startPoint(startPoint),
      |                                          ^~~~~~~~~~~
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:96:54: error: ‘startPoint’ was not declared in this scope
   96 |                                          _startPoint(startPoint),
      |                                                      ^~~~~~~~~~
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:97:42: error: class ‘TrajectoryBase<DataType>’ does not have any field named ‘_endPoint’
   97 |                                          _endPoint(endPoint),
      |                                          ^~~~~~~~~
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:97:52: error: ‘endPoint’ was not declared in this scope
   97 |                                          _endPoint(endPoint),
      |                                                    ^~~~~~~~
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:98:53: error: ‘startTime’ was not declared in this scope; did you mean ‘_startTime’?
   98 |                                          _startTime(startTime),
      |                                                     ^~~~~~~~~
      |                                                     _startTime
/noetic/software_robot_library/Trajectory/include/TrajectoryBase.h:99:51: error: ‘endTime’ was not declared in this scope; did you mean ‘_endTime’?
   99 |                                          _endTime(endTime),
[... cut ...]
/noetic/software_robot_library/Trajectory/include/Polynomial.h:236:21: error: ‘struct State<float>’ has no member named ‘acceleration’
  236 |     if(j > 1) state.acceleration(i) += (j-1)*j*this->_coefficients(i,j)*pow(time,j-2);
      |               ~~~~~~^~~~~~~~~~~~
make[2]: *** [Trajectory/CMakeFiles/Trajectory.dir/build.make:63: Trajectory/CMakeFiles/Trajectory.dir/test/polynomial_test.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:313: Trajectory/CMakeFiles/Trajectory.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

Woolfrey, 2024-07-12:
Thanks @anubhav-dogra! I will look in to it.

I ended up getting rid of most of the templates since it's slowing down compile time, so this might help circumvent a lot of problems.

Woolfrey, 2024-07-12:
@ssutjipto, @anubhav-dogra:

Here's what I've found out:
1. Eigen 3.4 is the latest version, which supports the `Eigen::Vector<T,Eigen::Dynamic>` template object instantiation.
2. Eigen 3.4 must be _manually_ installed on Ubuntu 20.04. Using `sudo apt install libeigen3-dev` will not work.

Here are the steps for manual installation:

[image]

anubhav-dogra, 2024-07-13:
Hi @Woolfrey 
Yup, this worked!

however, libeigen3-dev remove command also removed many ROS based packages which might be required to reinstalled when required.

anubhav-dogra, 2024-07-19:
I cloned this repo again, tried compiling it.
However this time the error is as following. 
```
[  4%] Building CXX object Math/CMakeFiles/Math.dir/src/MathFunctions.cpp.o
In file included from /home/vscode/stuff_ws/src/software_robot_library/Math/src/MathFunctions.cpp:8:
/home/vscode/stuff_ws/src/software_robot_library/Math/include/MathFunctions.h:72:6: error: ‘vector’ in namespace ‘std’ does not name a template type
   72 | std::vector<double>
      |      ^~~~~~
In file included from /home/vscode/stuff_ws/src/software_robot_library/Math/src/MathFunctions.cpp:8:
/home/vscode/stuff_ws/src/software_robot_library/Math/include/MathFunctions.h:13:1: note: ‘std::vector’ is defined in header ‘<vector>’; did you forget to ‘#include <vector>’?
   12 | #include <iostream>
  +++ |+#include <vector>
   13 | 
/home/vscode/stuff_ws/src/software_robot_library/Math/src/MathFunctions.cpp:187:6: error: ‘vector’ in namespace ‘std’ does not name a template type
  187 | std::vector<double>
      |      ^~~~~~
/home/vscode/stuff_ws/src/software_robot_library/Math/src/MathFunctions.cpp:9:1: note: ‘std::vector’ is defined in header ‘<vector>’; did you forget to ‘#include <vector>’?
    8 | #include <MathFunctions.h>
  +++ |+#include <vector>
    9 | 
make[2]: *** [Math/CMakeFiles/Math.dir/build.make:63: Math/CMakeFiles/Math.dir/src/MathFunctions.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:185: Math/CMakeFiles/Math.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

Woolfrey, 2024-07-19:
I did `make clean` and rebuilt the package:

[image]

Then ran `sudo make install`:

[image]

Over in [the test repository](https://github.com/Woolfrey/testing_robot_library) I also did `make clean` then `make`:

[image]

The tests work just fine :thinking:

anubhav-dogra, 2024-07-19:
@Woolfrey 
Yes I checked again, 
had to add `#include <vector>` in `Mathfunction.h`, `Spine.h` and `TrapezoidalVelocity.h`
Also editted `Pose pose(xyz, AngleAxis(rpy(0),...` to `Pose pose(xyz, AngleAxis<double>(rpy(0),` in `KinematicTree.cpp`

Now devel branch is compiled. Cheers

Woolfrey, 2024-07-19:
@anubhav-dogra, have you forked the repository? Are you able to make a merge request with your changes?

Woolfrey, 2024-12-18:
Updated the instructions on the `README.md` to reflect this.

closed 2024-12-18 by Woolfrey