Woolfrey/software_robot_library #122: Test Cartesian velocity control algorithm

Woolfrey, 2024-05-28:
I'm currently working on a Cartesian velocity control test executable.

I am sometimes getting very nice results:

[image]

[image]

[image]

---
This one starts with a high joint velocity, which corresponds to a high initial orientation error of the endpoint:

[image]

[image]

There is obviously something wrong with the orientation error calculation as the initial pose of the robot is used as the start point of the trajectory generator (it should have zero error at the start).

On the plus side, the QP solver is obeying the joint limits very nicely!

Woolfrey, 2024-05-29:
I fixed the problems with the orientation error. It was inside the `CartesianTrajectory` class itself.

Here's an example of a numerical simulation with good results:

[image]

[image]

[image]
---
Here is a numerical simulation with "bad" results:

[image]

[image]

[image]

:+1: The good thing is, the QP solver is obeying the joint limits!
:-1: The output for singularity avoidance is unstable. We need to work on it.

I still need to test a 6DOF robot as well.

Woolfrey, 2024-05-30:
I tried numerous methods of singularity avoidance, including fancy methods of momentum, QR decomposition, etc. etc. I settled on 2 things:
1. Reducing the max joint acceleration in the `SerialLinkBase` class, and
2. Increasing the damping in a classical DLS for the `SerialKinematicControl` class.

Results are satisfactory, but not ideal:

[image]

[image]

It's hard to say what's realistic since I'm just running random numbers; I can't if the robot is in a feasible position, if the trajectories are sensible, etc etc.

But numerically it looks "good enough" compared to previous results.

I still want to try 1 more method before moving on and settling with this for short-term implementation.

ssutjipto, 2024-06-02:
Results look reasonable and your plan sounds good. 

Also, do you have the python scripts for plotting in another repository?

Woolfrey, 2024-06-02:
@ssutjipto, the Python scrips are in the test folder:

https://github.com/Woolfrey/software_robot_library/tree/control/Control/test

Woolfrey, 2024-06-03:
Here are results using the UR5 model:

[image]

[image]

[image]

--- 

Here is a "bad" case where the robot hits a singularity:

[image]

[image]

[image]

The QP solver satisfied the constraint on manipulability nicely.

(I've also seen some crazy / unstable results, but its hard to ascertain if these are sensible without doing a proper 3D simulation.)